conservationtechlab/dencam_hardware #23: Camera needs to be flipped: new design for supporting it must be done

iingram, 2024-12-11:
The plan post Churchill visit is to go ahead and redesign around the camera being in the more conventional position i.e. dome down.  This means re-engineering/shopping associated with the tripod and associated assemblies/parts, hopefully retaining as much design from how things were but being ready for a lot of changes.  This is a high priority and there's also need (perhaps this should be captured in another issue) to test the aiming when the camera is in this position but also angled up 10, 15, maybe 25 degrees as will be necessary to look up the hillside.

sumi1194, 2024-12-12:
An option for a tripod style that would allow the camera to be mounted upside down without the view being obstructed by the legs: https://www.kentfaith.com/KF09.086_k-f-sa254t1-lightweight-compact-tripod-for-canon-nikon-camera-professional-dslr-photography-93inch-aluminium?srsltid=AfmBOooa4HecxVwPOih8EYBoqIR319HUgl13O4dxHDOVg1JykfL-SGKHyiM&gQT=1

At first glace, it seems like we could keep a similar (if not the same) mounting structure as well.

tonytonyhsiao, 2024-12-25:
Update: The CAD design was completed on 12/18/2024 and is ready for 3D printing. However, as of 12/24/2024, the 3D printer keeps crashing during the printing process.

kgarwoodsdzwa, 2025-01-10:
[image]
[image]
Current status of CAD printed mount. Holds the camera well and secures nicely to the tripod swivel with 3 bolts, allowing full range of motion of the swivel stand. There is a chamber on the top that holds the wires from the camera inside with a through-hole for the ethernet cable used to connect the camera to the box. Will likely print 1 more design with some slight modifications but so far works as proof of concept.

kgarwoodsdzwa, 2025-01-10:
[image]
[image]

tonytonyhsiao, 2025-01-14:
Updates to modify the wiper design to prevent interference between them.

[image]

iingram, 2025-02-27:
This design and build was taken to Svalbard and tested a little by the SDZWA "field team" indoors and just outside the building in that context and will be handed over to the PBI team as part of the whole Mimir system handoff that will progress to further testing outdoors.  We will close this issue as done, tracking any changes that may be revealed to be necessary as that testing progresses as separate issues.