conservationtechlab/dencam_hardware #24: Cyclops: Smaller batteries for shipping regulations

sumi1194, 2024-12-12:
Shipping regulations make transporting the current LTO batteries from XS power cost prohibitive. Maybe need to look into smaller battery types that fit within the limitations.

iingram, 2024-12-18:
Also check key airlines (eg SAS) regulations to make sure the 100Wh thing pertains.

KieranMcIver, 2024-12-22:
- XS Power responded. Once I can provide a shipping address they will look into options for shipping the batteries from their facility. I reached out to Christian for the address. 

- Robin Wainwright responded as well. Please review his response (I have attached his message below) and let me know if we should continue pursuing this option. Could you also provide a photo of the battery compartment on the cam setup? I will pass it on to Robin. 
- 
[image]

iingram, 2024-12-22:
Glad to hear about the result from XS power.  I figure this is clear but we should definitely keep going on that tack and hopefully it works out.

Re: Robin: my basic feeling is I would like to keep going on this additional tack to a) possibly have a backup solution depending on how it goes with XS b) have a source for 100Wh LTO batteries for other such projects and c) have a custom vendor in our vendor list who can do this kind of thing as it feels as if a custom need for LTO batteries is likely to come up again. I would, though, like to know what the total price to get ~400-500Wh will be.  I don't unfortunately have the time to continue to try to interpret Robin's message to suss that out right at this instant.  Could you try to suss that out?  Ie is he saying $750 for the whole final build that gets us ~500Wh or $750 for each 92Wh unit or what? I think the answer can be puzzled out of his text with a careful read. My current read is that it is $750 for each 240Wh unit which is the same price as each 230Wh battery we are buying now so it's fine.

Re: photo of the battery compartment, I will see what I have.  If I can't find anything, @kgarwoodsdzwa can you get one and send it tomorrow?

iingram, 2024-12-22:
[image]

iingram, 2024-12-22:
Found the above photo from Oct 20.  New build has slight differences that @tonytonyhsiao can describe if necessary. But this photo is probably sufficient to send Robin as we are still using the same pelican box and the amount of space reserved for the batteries is the same or even marginally increased. Plus I think this photo answers the question Robin is currently actually asking.

tonytonyhsiao, 2024-12-22:
[image]
[image]

tonytonyhsiao, 2024-12-22:
The photos attached show the latest design updates. As long as the dimensions of the new battery remain within the combined dimensions of two Titan8s, the current design should work fine.

KieranMcIver, 2024-12-22:
Thank you for the info and the photos. Robin is also asking for the charge controller details. I can't quite make out all the information from the photos. Is there a model number to go along with SunSaver MPPT?

tonytonyhsiao, 2024-12-22:
It's a Morningstar SunSaver SS-MPPT-15L MPPT Charge Controller

KieranMcIver, 2025-01-02:
I received the following quote from Robin. Let me know what you think.

He said the baseplate in the rendering represents the footprint of the existing pair of Titan8 RSV-5S's

[Polar Bear LTO quote.pdf](https://github.com/user-attachments/files/18289274/Polar.Bear.LTO.quote.pdf)

[image]

kgarwoodsdzwa, 2025-01-03:
This looks pretty good, was he able to confirm that our solar charge controller will be compatible if we adjust the settings? If so, I think we could move forward with ordering these. It looks like the nominal voltage is .5 V below our current set-up but I do think the solar charge controller we're using can handle it. If he would be able to provide insight on the necessary setting changes that we'd need to make to our solar charge controller (currently expecting nominal voltage of 12V) that would be helpful.

KieranMcIver, 2025-01-06:
Here is a txt thread from Robin addressing some of your concerns. Let me know if you would like me to move forward and place the order. I will require a shipping address for your location, and he will add the cost to the estimate.

 [image]