conservationtechlab/dencam_hardware #27: Build second Mimir system

iingram, 2024-12-19:
Build a second Mimir system now that the first is in Churchill.  This will be the one that we bring to Svalbard to ideally deploy at a den site.  This work is well underway, just wanted to capture it.

tonytonyhsiao, 2024-12-25:
Update: Completed as of 12/25/2024 and ready for review.

tonytonyhsiao, 2024-12-25:
[image]

tonytonyhsiao, 2024-12-25:
Still waiting for the PTZ camera connector to arrive; once it does, the hardware side will be completely done.

tonytonyhsiao, 2024-12-31:
Completed assembling the PTZ camera connector plug for Ethernet cables.

tonytonyhsiao, 2024-12-31:
[image]

iingram, 2025-02-27:
This new build (we could call it prototype number #002, with #001 being the one in Churchill) has, as of today, been fully prepared and run.  Full mechanical integration was achieved (well, almost achieved: see some items mixed in below), but not fully reviewed, a few weeks back I believe.  The remaining steps, executed over the passed few days, involved, amongst other things:

- configuring the settings in the Hikvision camera to what they need to be to work with the system (of particular importance was turning ONVIF communication on)
- getting the firmware to properly start up on boot (this is best done as a `systemd` service but is currently resolved for this unit as a `autostart` application).
- setting the proper DIP switch on the charge controller to allow communication with the firmware so that it can read the charge controller data
- attaching an SD card reader to one of the Pi's USB ports (to complete the system and test it but also because there's a crashing bug (documented elsewhere) if no external media has ever been attached to the PI OS running on the Pi).
- testing that all the functionality of a Mimir Dencam is working on the unit (tests that won't pass if the above things haven't been done)
- adding code to the firmware that addresses the camera being inverted from prior orientation
- resolving the two zoom-related bugs when controlling the PS4 controller

Things that were left out of the build that should have been there:
- above mentioned USB SD card reader/writer
- USB adapter that angles the port ninety degree "up" from Pi.

I will move this issue to Review and then we can look over it together when we do a hotwash next week.

closed 2025-03-06 by iingram