conservationtechlab/dencam_hardware #43: Make a third build (Protoype #003) of the Mimir System

iingram, 2025-02-28:
The first build went to Churchill and is with Kieran.  The second build went to Svalbard and is with Christian and Hilde.  Now we need a third build to continue engineering at the CTL.  Because of other priorities and the chance that we might add/swap features in response to the results of the 2025 deployment etc., completing this build is probably something we should target for the end of the Q3 2025.

kgarwoodsdzwa, 2025-03-26:
purchased needed parts to build second mimir system today (mar 26), need to search for a replacement part for the footpads as mentioned in an issue in dencam_hardware repo.

kgarwoodsdzwa, 2025-03-26:
still need to purchase metal assembly parts