conservationtechlab/dencam_hardware #45: Make totally stripped down POC of Starlink on a Pi to get the connectivity we want

iingram, 2025-03-31:
i.e. things like can we set things up so we don't need the most expensive tier of Starlink service

kgarwoodsdzwa, 2025-04-01:
Weatherproof box
Raspberry pi 5
RTC Battery
ethernet cable out, pi to starlink 
ac to dc 12v converter
power out to starlink, 12v dc
12v to 5v stepdown converter for pi
assigned to bob april 1

kgarwoodsdzwa, 2025-04-15:
deployed in oll, still waiting for the starlink to make a connection but it's running a script to email me the public ip every 12 hours and upon reboots. it's still connected to wifi, but i set the metrics to prioritize ethernet once it comes on so there's a chance it will be able to send me the starlink ip once it comes on. there's a chance it won't but its a good start. hardware all set but possible the starlink needs more setup/ or our suspicions are confirmed and we need to splurge for the static ip plan. if it does prioritize eth and is able to send an email though, it would mean we don't need static for an outbound connection at least. the send email script keeps logs so ill be able to check the logs to see what happened. could make the logs more verbose to also add what it thinks it's eth and wan connections are from network manager

[image]